makerbase-mks/MKS-Robin-Nano-V3.X #149: mks robin nano v3.1 jumper caps position

pulsar1955, 2023-03-20:
good morning, I need help, I bought a mks robin nano v3.1 and I should mount the drv 8825 drivers, but due to my inexperience I don't know how to set the jumper caps, I haven't found any info on this version of the mother board the other versions have a different layout maybe a photo, a drawing ps the power is 3.3v or 5v??
Thanks in advance
Robert

MKS-hosiey, 2023-03-23:
[image]

MKS-hosiey, 2023-03-23:
This is 16 subdivisions

pulsar1955, 2023-03-24:
Thanks MKS-hosiery, it was the confirmation I was waiting for, the position of the jumpers seemed strange to me, and before doing any damage, I'd better get one more opinion, particularly authoritative like yours.
thank you and wish you a happy easter