makerbase-mks/MKS-Robin-Nano-V3.X #161: robin nano v3 - Printrun (pronterface) serial connection problem

bart-tech3d, 2023-07-08:
Hi, I have a problem connecting this board via USB to a PC and controlling it via printrun. How to proceed correctly? This board is probably no longer for the CH340 Driver?
I am using marlin 2.1.
The only thing I was able to achieve was to display the files on the USB flash drive inserted in the mks board on the PC.
Thank you

lulanascimento, 2023-09-09:
@bart-tech3d 
Did I manage to solve your problem?
I have the same problem, the white screen when turning on, I can't update the firmware or connect using the serial port.

bart-tech3d, 2023-09-10:
Hello, 
no, I haven't solved the problem yet. Unfortunately, no one responded to this. As for the white screen, try writing 
#define TFT_DRIVER ST7796 in pins_MKS_ROBIN_NANO_V3_common.h

Zaquron, 2024-01-25:
i have the same problem with USB connection. 
It worked for a few minutes at first.
After a few reboots it no longer comuunicates through USB.
Windows sees serial port but no communication.
Tried a few different verisons of Marlin to no avail.

YHamud, 2024-01-31:
I only managed to connect by setting Serial Port 0 -1 in configuration.h.

bart-tech3d, 2024-02-01:
do You can share Your configuration.h file please?

Zaquron, 2024-02-01:
My issue seems to be something wrong with power from usb. As soon as 12V is supplied to the board and usb power jumper is removed everything works just fine.

YHamud, 2024-02-02:
> do You can share Your configuration.h file please?

If you mean me, yes, for sure. It's attached.
[Configuration.zip](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/files/14133519/Configuration.zip)