makerbase-mks/MKS-Robin-Nano-V3.X #163: Mks robin nano v3.1 vscode marlin derleme hatası

caliskanufuk06, 2023-08-12:
Mks robin nano v3.1 aldım. Hiçbir yerde nasıl yükleyeceğimizle ilgili bilgi yok. Vscode da yüklerden hata veriyor.  Üretici kaynak dosyalarıyla yükleme yapılamıyor. Marlin 2.1.2 dışında marlin/scr/core/bords kısmında v3.1 tanımlı değil. Mks gitHub kaynakları işe yaramıyor
1. MKS Robin Nano v3.0 operasyon talimatları:
https://blog.csdn.net/gjy_skyblue/article/details
/118754258?spm=1001.2014.3001.5501
2. MKS Robin Nano V3 Github:
https://github.com/makerbase-mks/MKS-Robin
-Nano-V3.X
3. Firmware kaynağı: https://github.com
/makerbase-mks/Mks-Robin-Nano-Marlin2.0
-Firmware

Lütfen acil yardım!

savasoskan, 2023-10-09:
merhaba sorunu çözebildiniz mi ? nasıl çözdünüz ?

caliskanufuk06, 2023-10-09:
Merhaba çözdüm. Sorun çok basitmiş. Vscode'da ben hata çıkarsa diye her ayardan sonra tek tek derleme yapıp kaydederek ilerliyordum. 
Marlinin en son sürümü yükle. Vscode kullanarak ana kartı , mks robin nano v3.1 olarak tanıt. Devamında 3d yazıcınız ile ilgili istediğin ayarları yap. Tüm ayarları yapıp en son olarak derle. Sonra yükle. Ve şu videoya da bir bak.
https://youtube.com/clip/UgkxcJ7SrNqZ3imh6kNxCTQFHSCmcgBTF7IA?si=uYzCQtr0PbTtgrLS