makerbase-mks/MKS-Robin-Nano-V3.X #169: Hardware PWM not working for part cooling fan FAN_PIN PC14

wagnerism, 2023-11-04:
I have the fan connected to PC14 "FAN1" on the board.   In `pins_MKS_ROBIN_NANO_V3_common.h` FAN_PIN is set to PC14.

The part cooling fan does not move when set between 0-49%, then it runs at full speed when set between 50-100%

I enabled `FAN_SOFT_PWM` and it appears to work but it is very noisy.  According to https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/blob/main/README.md there are two hardware PWM fan ports.

How does one use the hardware PWM associated with pin PC14?

YHamud, 2024-01-31:
I'm facing the same problem with PC14 (fan0) on the TT SP5-V3, will try with PB1 (fan1). The main problem is that the printer used 2 high rpm part cooling fans and at  full spin they are really powerful, which drops a lot the nozzle temperature and may not be very good for layer adhesion.

YHamud, 2024-02-02:
Hear me out, I found a way around it.

Just make the Fan1 pin your Fan0. Go to the pins file (pins_MKS_ROBIN NANO_V3_common.h) and replace PC14 by PB1 on FAN0_PIN and put -1 on FAN1_PIN).
Then on the board just plug the part cooling fans on the fan socket beside the current one.

I tested and now both fans start as low as 20% (very low whining) and speed up to 100% without issues.

Let me know if you face any problem while doing that.

th00mas88, 2024-02-16:
@YHamud  

I have had the same issue.   

Done as described and i cannot make marlin as i had an error calling it FAN0_PIN  but i have made FAN1_PIN -1 

I have changed FAN_PIN to PB1 and PC14 neither work .

th00mas88, 2024-02-20:
following up - i was on a slightly older marlin

 FAN0_PIN i have swapped to PB1 and PC14  
 FAN1_PIN has been -1 . still not working

l-e-s, 2024-03-09:
I tried to swap fan pins as described above, also set PC14 as a hot1 fan to automatically start when the extruder becomes hot:
pins_MKS_ROBIN NANO_V3_common.h
```
#define FAN1_PIN                            PC14  // FAN
#define FAN0_PIN                            PB1   // FAN1
```
Configuration_adv.h
```
#define E0_AUTO_FAN_PIN FAN1_PIN
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 255   // 255 == full speed
```
Works well for me. Thank you very much for advice!

simpit303, 2024-04-02:
Thank you gents for your post here.
I can confirm that above method is working perfectly on my Robin Nano 3.0 with Marlin bugfix 2.1.x.
Now (Just like in Marlin 2.0.x) i have ability to control speed on second Cooler fan in TS35 screen menu ( 50% Speed and 100% Speed)
Extruder fan is working automaticaly just perfect. You don't hae to set in ...adv.h nothing for cooer fan just set E0_AUTO _FAN FAN1_PIN for extruder fan.