makerbase-mks/MKS-Robin-Nano-V3.X #176: Robin-Nano-V3.1 + MKS MINI12864 V3 Visual Studio Code builld error

Ignasbal, 2024-03-07:
I enable (#define MKS_MINI_12864_V3 enable: #define NEOPIXEL_LED)  in the Configuration.h like written in MKS [docs](https://github.com/makerbase-mks/MKS-MINI12864-V3?tab=readme-ov-file#firmware-config)

The error I get is: "MKS Robin Nano V3 doesn't support FSMC-based TFT displays."
Is there anything else I should add, enable?

[image]

killyane, 2024-04-12:
Did you find a solution? I have the same issue.
If yes, please help me out

jamesmackgyver, 2024-04-29:
Hi i have same problem.Thanks

Ignasbal, 2025-01-16:
My setup files.

[https://drive.google.com/drive/folders/1LvtYqM9Flx3OALy0XGOLy4X40VV-T4pv]

Hope it helps.

closed 2025-01-16 by Ignasbal